jbush001/NyuziToolchain #83: Investigate whole function vectorization

jbush001, 2017-06-24:
http://compilers.cs.uni-saarland.de/projects/wfv/#header4

This seems to be an alternative to POCL.

rkruppe, 2017-07-06:
Heads up, I've been working on this for a while (https://github.com/jbush001/NyuziToolchain/issues/44 was part of that). It'll be some more time before I have results to show for, but I intend to make it public asap.

rkruppe, 2017-08-22:
FWIW, the code is now public at https://github.com/rkruppe/spmd-vectorize. It's very much research-quality code at the moment, and there's a long list of known limitations that somewhat restrict the usefulness (most of which aren't challenging to add, I'm just knee deep in other stuff right now), but if you want to take a look, I've applied it to two of your hand-vectorized benchmarks, [mandelbrot](https://github.com/rkruppe/spmd-vectorize-benchmarks/blob/c73992878eb89e74e02def65643abac7e98d4250/mandelbrot/mandelbrot.cpp) and [hash](https://github.com/rkruppe/spmd-vectorize-benchmarks/blob/c73992878eb89e74e02def65643abac7e98d4250/hash/hash.cpp). (The former is competitive performance-wise, the latter is not for reasons that elude me.)

jbush001, 2017-08-22:
That's awesome!