jbush001/NyuziToolchain #106: An error occurs when running ninja build on centos7

donttal, 2019-11-02:
## Error

../lib/CodeGen/SelectionDAG/LegalizeTypes.cpp:1115:1: fatal error: error writing to /tmp/ccmwXfUA.s: No space left on device

../lib/CodeGen/AsmPrinter/AddressPool.cpp:77:1: fatal error: error writing to /tmp/cc1lUP51.s: No space left on device

../lib/CodeGen/AsmPrinter/DwarfCFIException.cpp:171:1: fatal error: error writing to /tmp/cctO5JJD.s: No space left on device

../lib/CodeGen/SelectionDAG/LegalizeFloatTypes.cpp:2214:1: fatal error: error writing to /tmp/ccAgAjez.s: No space left on device

../lib/CodeGen/SelectionDAG/SelectionDAGAddressAnalysis.cpp:295:1: fatal error: error writing to /tmp/ccI05tf0.s: No space left on device

../lib/CodeGen/AsmPrinter/DebugLocStream.cpp:45:1: fatal error: error writing to /tmp/ccNpBg5J.s: No space left on device

../lib/CodeGen/SelectionDAG/SelectionDAGPrinter.cpp:304:1: fatal error: error writing to /tmp/ccUhA1Fy.s: No space left on device

../lib/CodeGen/AsmPrinter/DebugHandlerBase.cpp:335:1: fatal error: error writing to /tmp/cc7OWmiy.s: No space left on device

../lib/CodeGen/AsmPrinter/AccelTable.cpp:732:1: fatal error: error writing to /tmp/cctXSmYB.s: No space left on device

../lib/CodeGen/SelectionDAG/ScheduleDAGFast.cpp:804:1: fatal error: error writing to /tmp/ccB7nmha.s: No space left on device

../lib/CodeGen/SelectionDAG/SelectionDAGISel.cpp:3889:30: fatal error: error writing to /tmp/cc5qbSnw.s: No space left on device

 char SelectionDAGISel::ID = 0;

../lib/CodeGen/SelectionDAG/SelectionDAGBuilder.cpp:10751:1: fatal error: error writing to /tmp/ccfFxfGz.s: No space left on device

#### How do I need to modify it?

jbush001, 2019-11-02:
It seems to be reporting that your disk is full: "No space left on device." Are you running on a virtual machine?

donttal, 2019-11-02:
> It seems to be reporting that your disk is full: "No space left on device." Are you running on a virtual machine?

I am running on the lab server, not a virtual machine.

jbush001, 2019-11-02:
You can type the command 'df' in a terminal to determine how much space is left.

donttal, 2019-11-03:
## 这个是我运行df -hl命令 显示的结果
[root@localhost build]# df -hl
Filesystem               Size  Used Avail Use% Mounted on
/dev/mapper/centos-root   50G   50G  252M 100% /
devtmpfs                  32G     0   32G   0% /dev
tmpfs                     32G  6.0M   32G   1% /dev/shm
tmpfs                     32G   66M   32G   1% /run
tmpfs                     32G     0   32G   0% /sys/fs/cgroup
/dev/sda2               1014M  174M  841M  18% /boot
/dev/sda1                200M   11M  189M   6% /boot/efi
/dev/mapper/centos-home  754G  141G  613G  19% /home
tmpfs                    6.3G   52K  6.3G   1% /run/user/0
## When I jump to the tmp directory and run du -sh
[root@localhost tmp]# du -sh
1.7M    .
### what should I do

jbush001, 2019-11-03:
What is the path to the directory you are trying to build in?

donttal, 2019-11-03:
> What is the path to the directory you are trying to build in?
### Do you mean the installation path of NyzuiToolchain?
/root/NyuziProcessor/tools/NyuziToolchain/build

jbush001, 2019-11-03:
I would suggest building it underneath your home directory. That is, clone the NyuziProcessor repo there. It appears you are trying to build on the root partition, which doesn't have space left (and isn't a great place to put this kind of stuff anyway).

donttal, 2019-11-03:
> 我建议将其构建在主目录下。也就是说，在此处克隆NyuziProcessor存储库。看来您正在尝试在根分区上构建，该分区没有剩余空间（无论如何也不是放置此类内容的好地方）。
I logged in to the server as the root user and its default HOME directory is /root. Should I log in with another account and enter the /home/usr directory?

jbush001, 2019-11-03:
Yes, I would recommend using a regular account.

donttal, 2019-11-07:
> 是的，我建议您使用普通帐户。
**When I run the ninja command as a normal user login, the following problem occurs.**
/opt/rh/devtoolset-6/root/usr/libexec/gcc/x86_64-redhat-linux/6.3.1/ld: /usr/local/lib/libpython3.7m.a(abstract.o): relocation R_X86_64_32S against symbol `_Py_NotImplementedStruct' can not be used when making a shared object; recompile with -fPIC
/opt/rh/devtoolset-6/root/usr/libexec/gcc/x86_64-redhat-linux/6.3.1/ld: /usr/local/lib/libpython3.7m.a(boolobject.o): relocation R_X86_64_32S against symbol `_Py_FalseStruct' can not be used when making a shared object; recompile with -fPIC
/opt/rh/devtoolset-6/root/usr/libexec/gcc/x86_64-redhat-linux/6.3.1/ld: /usr/local/lib/libpython3.7m.a(bytearrayobject.o): relocation R_X86_64_32 against symbol `_PyByteArray_empty_string' can not be used when making a shared object; recompile with -fPIC
/opt/rh/devtoolset-6/root/usr/libexec/gcc/x86_64-redhat-linux/6.3.1/ld: /usr/local/lib/libpython3.7m.a(bytesobject.o): relocation R_X86_64_32 against symbol `_Py_NoneStruct' can not be used when making a shared object; recompile with -fPIC
/opt/rh/devtoolset-6/root/usr/libexec/gcc/x86_64-redhat-linux/6.3.1/ld: /usr/local/lib/libpython3.7m.a(call.o): relocation R_X86_64_32S against symbol `PyFunction_Type' can not be used when making a shared object; recompile with -fPIC
/opt/rh/devtoolset-6/root/usr/libexec/gcc/x86_64-redhat-linux/6.3.1/ld: /usr/local/lib/libpython3.7m.a(classobject.o): relocation R_X86_64_32 against `.rodata.str1.1' can not be used when making a shared object; recompile with -fPIC
/opt/rh/devtoolset-6/root/usr/libexec/gcc/x86_64-redhat-linux/6.3.1/ld: /usr/local/lib/libpython3.7m.a(exceptions.o): relocation R_X86_64_32S against symbol `_Py_NoneStruct' can not be used when making a shared object; recompile with -fPIC
/opt/rh/devtoolset-6/root/usr/libexec/gcc/x86_64-redhat-linux/6.3.1/ld: /usr/local/lib/libpython3.7m.a(fileobject.o): relocation R_X86_64_32 against symbol `_Py_NoneStruct' can not be used when making a shared object; recompile with -fPIC
/opt/rh/devtoolset-6/root/usr/libexec/gcc/x86_64-redhat-linux/6.3.1/ld: /usr/local/lib/libpython3.7m.a(floatobject.o): relocation R_X86_64_32 against `.rodata.str1.1' can not be used when making a shared object; recompile with -fPIC
/opt/rh/devtoolset-6/root/usr/libexec/gcc/x86_64-redhat-linux/6.3.1/ld: /usr/local/lib/libpython3.7m.a(frameobject.o): relocation R_X86_64_32 against symbol `_Py_NoneStruct' can not be used when making a shared object; recompile with -fPIC
/opt/rh/devtoolset-6/root/usr/libexec/gcc/x86_64-redhat-linux/6.3.1/ld: /usr/local/lib/libpython3.7m.a(funcobject.o): relocation R_X86_64_32S against symbol `PyCode_Type' can not be used when making a shared object; recompile with -fPIC
/opt/rh/devtoolset-6/root/usr/libexec/gcc/x86_64-redhat-linux/6.3.1/ld: /usr/local/lib/libpython3.7m.a(iterobject.o): relocation R_X86_64_32 against symbol `_Py_NoneStruct' can not be used when making a shared object; recompile with -fPIC
/opt/rh/devtoolset-6/root/usr/libexec/gcc/x86_64-redhat-linux/6.3.1/ld: /usr/local/lib/libpython3.7m.a(listobject.o): relocation R_X86_64_32 against symbol `_Py_NoneStruct' can not be used when making a shared object; recompile with -fPIC
/opt/rh/devtoolset-6/root/usr/libexec/gcc/x86_64-redhat-linux/6.3.1/ld: /usr/local/lib/libpython3.7m.a(longobject.o): relocation R_X86_64_32 against `.rodata.str1.1' can not be used when making a shared object; recompile with -fPIC
/opt/rh/devtoolset-6/root/usr/libexec/gcc/x86_64-redhat-linux/6.3.1/ld: /usr/local/lib/libpython3.7m.a(dictobject.o): relocation R_X86_64_32S against `.text.hot' can not be used when making a shared object; recompile with -fPIC
/opt/rh/devtoolset-6/root/usr/libexec/gcc/x86_64-redhat-linux/6.3.1/ld: /usr/local/lib/libpython3.7m.a(memoryobject.o): relocation R_X86_64_32 against `.rodata.str1.8' can not be used when making a shared object; recompile with -fPIC
/opt/rh/devtoolset-6/root/usr/libexec/gcc/x86_64-redhat-linux/6.3.1/ld: /usr/local/lib/libpython3.7m.a(methodobject.o): relocation R_X86_64_32 against symbol `_Py_NoneStruct' can not be used when making a shared object; recompile with -fPIC
/opt/rh/devtoolset-6/root/usr/libexec/gcc/x86_64-redhat-linux/6.3.1/ld: /usr/local/lib/libpython3.7m.a(moduleobject.o): relocation R_X86_64_32 against `.rodata.str1.8' can not be used when making a shared object; recompile with -fPIC
/opt/rh/devtoolset-6/root/usr/libexec/gcc/x86_64-redhat-linux/6.3.1/ld: /usr/local/lib/libpython3.7m.a(object.o): relocation R_X86_64_32 against `.rodata.str1.1' can not be used when making a shared object; recompile with -fPIC
/opt/rh/devtoolset-6/root/usr/libexec/gcc/x86_64-redhat-linux/6.3.1/ld: /usr/local/lib/libpython3.7m.a(obmalloc.o): relocation R_X86_64_32 against `.rodata.str1.8' can not be used when making a shared object; recompile with -fPIC
/opt/rh/devtoolset-6/root/usr/libexec/gcc/x86_64-redhat-linux/6.3.1/ld: /usr/local/lib/libpython3.7m.a(capsule.o): relocation R_X86_64_32 against `.rodata.str1.1' can not be used when making a shared object; recompile with -fPIC
/opt/rh/devtoolset-6/root/usr/libexec/gcc/x86_64-redhat-linux/6.3.1/ld: /usr/local/lib/libpython3.7m.a(rangeobject.o): relocation R_X86_64_32 against symbol `_Py_NoneStruct' can not be used when making a shared object; recompile with -fPIC
[... cut ...]
/opt/rh/devtoolset-6/root/usr/libexec/gcc/x86_64-redhat-linux/6.3.1/ld: /usr/local/lib/libpython3.7m.a(getcompiler.o): relocation R_X86_64_32 against `.rodata.str1.8' can not be used when making a shared object; recompile with -fPIC
/opt/rh/devtoolset-6/root/usr/libexec/gcc/x86_64-redhat-linux/6.3.1/ld: final link failed: Nonrepresentable section on output
collect2: error: ld returned 1 exit status
[3636/3645] Linking CXX executable bin/lldb-test
ninja: build stopped: subcommand failed.
**How should I solve it?**

jbush001, 2019-11-07:
Try going into tools/NyuziToolchain/tools/CMakeLists.txt and commenting out the following line (put a # in front of it):

    add_llvm_external_project(lldb)

donttal, 2019-11-08:
> Try going into tools/NyuziToolchain/tools/CMakeLists.txt and commenting out the following line (put a # in front of it):
> 
> ```
> add_llvm_external_project(lldb)
> ```

### I did what you said, re-running the cmake command and then running ninja-build, but the same error still happened.
/opt/rh/devtoolset-6/root/usr/libexec/gcc/x86_64-redhat-linux/6.3.1/ld: /usr/local/lib/libpython3.7m.a(abstract.o): relocation R_X86_64_32S against symbol `_Py_NotImplementedStruct' can not be used when making a shared object; recompile with -fPIC
/opt/rh/devtoolset-6/root/usr/libexec/gcc/x86_64-redhat-linux/6.3.1/ld: /usr/local/lib/libpython3.7m.a(boolobject.o): relocation R_X86_64_32S against symbol `_Py_FalseStruct' can not be used when making a shared object; recompile with -fPIC
/opt/rh/devtoolset-6/root/usr/libexec/gcc/x86_64-redhat-linux/6.3.1/ld: /usr/local/lib/libpython3.7m.a(bytearrayobject.o): relocation R_X86_64_32 against symbol `_PyByteArray_empty_string' can not be used when making a shared object; recompile with -fPIC
/opt/rh/devtoolset-6/root/usr/libexec/gcc/x86_64-redhat-linux/6.3.1/ld: /usr/local/lib/libpython3.7m.a(bytesobject.o): relocation R_X86_64_32 against symbol `_Py_NoneStruct' can not be used when making a shared object; recompile with -fPIC
/opt/rh/devtoolset-6/root/usr/libexec/gcc/x86_64-redhat-linux/6.3.1/ld: /usr/local/lib/libpython3.7m.a(call.o): relocation R_X86_64_32S against symbol `PyFunction_Type' can not be used when making a shared object; recompile with -fPIC
/opt/rh/devtoolset-6/root/usr/libexec/gcc/x86_64-redhat-linux/6.3.1/ld: /usr/local/lib/libpython3.7m.a(classobject.o): relocation R_X86_64_32 against `.rodata.str1.1' can not be used when making a shared object; recompile with -fPIC
/opt/rh/devtoolset-6/root/usr/libexec/gcc/x86_64-redhat-linux/6.3.1/ld: /usr/local/lib/libpython3.7m.a(exceptions.o): relocation R_X86_64_32S against symbol `_Py_NoneStruct' can not be used when making a shared object; recompile with -fPIC
/opt/rh/devtoolset-6/root/usr/libexec/gcc/x86_64-redhat-linux/6.3.1/ld: /usr/local/lib/libpython3.7m.a(fileobject.o): relocation R_X86_64_32 against symbol `_Py_NoneStruct' can not be used when making a shared object; recompile with -fPIC
/opt/rh/devtoolset-6/root/usr/libexec/gcc/x86_64-redhat-linux/6.3.1/ld: /usr/local/lib/libpython3.7m.a(floatobject.o): relocation R_X86_64_32 against `.rodata.str1.1' can not be used when making a shared object; recompile with -fPIC
/opt/rh/devtoolset-6/root/usr/libexec/gcc/x86_64-redhat-linux/6.3.1/ld: /usr/local/lib/libpython3.7m.a(frameobject.o): relocation R_X86_64_32 against symbol `_Py_NoneStruct' can not be used when making a shared object; recompile with -fPIC
/opt/rh/devtoolset-6/root/usr/libexec/gcc/x86_64-redhat-linux/6.3.1/ld: /usr/local/lib/libpython3.7m.a(funcobject.o): relocation R_X86_64_32S against symbol `PyCode_Type' can not be used when making a shared object; recompile with -fPIC
/opt/rh/devtoolset-6/root/usr/libexec/gcc/x86_64-redhat-linux/6.3.1/ld: /usr/local/lib/libpython3.7m.a(iterobject.o): relocation R_X86_64_32 against symbol `_Py_NoneStruct' can not be used when making a shared object; recompile with -fPIC
/opt/rh/devtoolset-6/root/usr/libexec/gcc/x86_64-redhat-linux/6.3.1/ld: /usr/local/lib/libpython3.7m.a(listobject.o): relocation R_X86_64_32 against symbol `_Py_NoneStruct' can not be used when making a shared object; recompile with -fPIC
/opt/rh/devtoolset-6/root/usr/libexec/gcc/x86_64-redhat-linux/6.3.1/ld: /usr/local/lib/libpython3.7m.a(longobject.o): relocation R_X86_64_32 against `.rodata.str1.1' can not be used when making a shared object; recompile with -fPIC
/opt/rh/devtoolset-6/root/usr/libexec/gcc/x86_64-redhat-linux/6.3.1/ld: /usr/local/lib/libpython3.7m.a(dictobject.o): relocation R_X86_64_32S against `.text.hot' can not be used when making a shared object; recompile with -fPIC
/opt/rh/devtoolset-6/root/usr/libexec/gcc/x86_64-redhat-linux/6.3.1/ld: /usr/local/lib/libpython3.7m.a(memoryobject.o): relocation R_X86_64_32 against `.rodata.str1.8' can not be used when making a shared object; recompile with -fPIC
/opt/rh/devtoolset-6/root/usr/libexec/gcc/x86_64-redhat-linux/6.3.1/ld: /usr/local/lib/libpython3.7m.a(methodobject.o): relocation R_X86_64_32 against symbol `_Py_NoneStruct' can not be used when making a shared object; recompile with -fPIC
/opt/rh/devtoolset-6/root/usr/libexec/gcc/x86_64-redhat-linux/6.3.1/ld: /usr/local/lib/libpython3.7m.a(moduleobject.o): relocation R_X86_64_32 against `.rodata.str1.8' can not be used when making a shared object; recompile with -fPIC
/opt/rh/devtoolset-6/root/usr/libexec/gcc/x86_64-redhat-linux/6.3.1/ld: /usr/local/lib/libpython3.7m.a(object.o): relocation R_X86_64_32 against `.rodata.str1.1' can not be used when making a shared object; recompile with -fPIC
/opt/rh/devtoolset-6/root/usr/libexec/gcc/x86_64-redhat-linux/6.3.1/ld: /usr/local/lib/libpython3.7m.a(obmalloc.o): relocation R_X86_64_32 against `.rodata.str1.8' can not be used when making a shared object; recompile with -fPIC
/opt/rh/devtoolset-6/root/usr/libexec/gcc/x86_64-redhat-linux/6.3.1/ld: /usr/local/lib/libpython3.7m.a(capsule.o): relocation R_X86_64_32 against `.rodata.str1.1' can not be used when making a shared object; recompile with -fPIC
/opt/rh/devtoolset-6/root/usr/libexec/gcc/x86_64-redhat-linux/6.3.1/ld: /usr/local/lib/libpython3.7m.a(rangeobject.o): relocation R_X86_64_32 against symbol `_Py_NoneStruct' can not be used when making a shared object; recompile with [... cut ...]
/opt/rh/devtoolset-6/root/usr/libexec/gcc/x86_64-redhat-linux/6.3.1/ld: /usr/local/lib/libpython3.7m.a(parser.o): relocation R_X86_64_32 against `.rodata.str1.8' can not be used when making a shared object; recompile with -fPIC
/opt/rh/devtoolset-6/root/usr/libexec/gcc/x86_64-redhat-linux/6.3.1/ld: /usr/local/lib/libpython3.7m.a(getcompiler.o): relocation R_X86_64_32 against `.rodata.str1.8' can not be used when making a shared object; recompile with -fPIC
/opt/rh/devtoolset-6/root/usr/libexec/gcc/x86_64-redhat-linux/6.3.1/ld: final link failed: Nonrepresentable section on output
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.

jbush001, 2019-11-08:
You might try deleting the whole build directory before re-running cmake to make sure. I have seen cases where it doesn't update when a project is removed. 

There is a known issue (in the README) about the LLDB debugger not building on systems that default to python3, but usually disabling it as I suggested previously fixes that.

donttal, 2019-11-08:
> 您可以尝试在重新运行cmake之前先删除整个构建目录，以确保确定。我已经看到了在删除项目后它不会更新的情况。
> 
> （自述文件中）有一个已知问题，关于LLDB调试器不在默认为python3的系统上构建，但通常会禁用它，因为我之前建议修复该问题。

## I did what you said, and I still reported the same error. The default Python version of the system is 2.7.5.
[root@localhost build]# python
Python 2.7.5 (default, Aug  7 2019, 00:51:29) 
[GCC 4.8.5 20150623 (Red Hat 4.8.5-39)] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> 

## And there is a new question, should I delete the extra symbol in the file？
[2249/3645] Building CXX object tools/clang/lib/StaticAnalyzer/Checkers/CMakeFiles/clangStaticAnalyzerCheckers.dir/Taint.cpp.o
../tools/clang/lib/StaticAnalyzer/Checkers/Taint.cpp:26:53: warning: extra ‘;’ [-Wpedantic]
                                        TaintTagType);

jbush001, 2019-11-09:
Hmmm... I'm not sure on the first one, I haven't seen that before. I'll try to do some research. 

On the extra symbol you mentioned, that's only a warning, there's no need to do anything.

jbush001, 2019-11-10:
Could you put the entire build output into a gist and post a link to it here?

donttal, 2019-11-11:
> gist
### here is
https://gist.github.com/donttal/f3804e6896cdd5f9f817a1a0a6179834
### thanks

jbush001, 2019-11-14:
Shoot, it's still trying to build LLDB. 

One more drastic thing to try is to go into the tools/NyuziToolchain directory and 'rm -rf tools/lldb'. Then in scripts/setup_tools.sh, comment out the two lines:

    git submodule init || fail "Error initializing submodules"
    git submodule update || fail "Error updating submodules"

You may need to then delete the whole build directory 'rm -rf build' and start over.

donttal, 2019-11-14:
> 射击，它仍在尝试构建LLDB。
> 
> 要尝试的另一种激烈方法是进入tools / NyuziToolchain目录和'rm -rf tools / lldb'。然后在scripts / setup_tools.sh中，注释掉两行：
> 
> ```
> git submodule init || fail "Error initializing submodules"
> git submodule update || fail "Error updating submodules"
> ```
> 
> 然后，您可能需要删除整个构建目录'rm -rf build'并重新开始。

**Thanks for your solution, the ninja-build command seems to have been running successfully, the output is as follows.** 

https://gist.github.com/donttal/5b20824c30ac231c98a3ccc202f212cf

**However, when I executed the next step sudo ninja-build install, an  new error occur. ninja no work to do and no install step for ‘builtins’. This does not seem to be a general environment-dependent installation problem.**

https://gist.github.com/donttal/66ccf5f2787d23774e02d7acf4881b7f

**Looking forward to your reply**

jbush001, 2019-11-15:
The 'no install step for builtins' thing is not a problem; that is informational. The thing that is breaking the build seems to be here:

```
CMake Error at tools/clang/tools/clang-check/cmake_install.cmake:41 (file):
  file INSTALL cannot copy file
  "/home/liuxing/NyuziProcessor/tools/NyuziToolchain/build/bin/clang-check"
  to "/usr/local/llvm-nyuzi/bin/clang-check".
```
You could run 'df' again. I suspect the problem is that you are running out of space on the root partition where it is trying to copy the files to. The LLVM installation takes a fair amount of space.

donttal, 2019-11-17:
[root@localhost liuxing]# ls -lh
total 4.0K
drwxr-xr-x. 11 root    root    4.0K Nov  7 10:45 NyuziProcessor
[root@localhost liuxing]# df
Filesystem              1K-blocks      Used Available Use% Mounted on
/dev/mapper/centos-root  52403200  52360364     42836 100% /
devtmpfs                 32828156         0  32828156   0% /dev
tmpfs                    32844528      6112  32838416   1% /dev/shm
tmpfs                    32844528    394752  32449776   2% /run
tmpfs                    32844528         0  32844528   0% /sys/fs/cgroup
/dev/sda2                 1038336    177416    860920  18% /boot
/dev/sda1                  204580     11240    193340   6% /boot/efi
/dev/mapper/centos-home 790223880 186055084 604168796  24% /home
tmpfs                     6568908        88   6568820   1% /run/user/0

> “没有内置步骤的安装步骤”是没有问题的。那是信息性的。破坏构建的东西似乎在这里：
> 
> ```
> CMake Error at tools/clang/tools/clang-check/cmake_install.cmake:41 (file):
>   file INSTALL cannot copy file
>   "/home/liuxing/NyuziProcessor/tools/NyuziToolchain/build/bin/clang-check"
>   to "/usr/local/llvm-nyuzi/bin/clang-check".
> ```
> 
> 您可以再次运行“ df”。我怀疑问题是您试图将文件复制到的根分区上的空间不足。LLVM安装占用大量空间。

#### here is
[root@localhost liuxing]# ls -lh
total 4.0K
drwxr-xr-x. 11 root    root    4.0K Nov  7 10:45 NyuziProcessor
[root@localhost liuxing]# df
Filesystem              1K-blocks      Used Available Use% Mounted on
/dev/mapper/centos-root  52403200  52360364     42836 100% /
devtmpfs                 32828156         0  32828156   0% /dev
tmpfs                    32844528      6112  32838416   1% /dev/shm
tmpfs                    32844528    394752  32449776   2% /run
tmpfs                    32844528         0  32844528   0% /sys/fs/cgroup
/dev/sda2                 1038336    177416    860920  18% /boot
/dev/sda1                  204580     11240    193340   6% /boot/efi
/dev/mapper/centos-home 790223880 186055084 604168796  24% /home
tmpfs                     6568908        88   6568820   1% /run/user/0

jbush001, 2019-11-17:
Yeah, looks like your root filesystem is full: 

    /dev/mapper/centos-root 52403200 52360364 42836 100% /

donttal, 2019-11-17:
> 是的，看来您的根文件系统已满：
> 
> ```
> /dev/mapper/centos-root 52403200 52360364 42836 100% /
> ```

This sounds like there is no other way to solve my problem?

jbush001, 2019-11-19:
There's always a way. :)

You could try running in a Docker container if you are familiar with that. The latest toolchain is here: https://hub.docker.com/repository/docker/jeffbush001/nyuzi-build

jbush001, 2019-11-22:
Another alternative, which might be easier, would be to edit the top level CMakeLists.txt to point to a different location. I think it will work just pointing to the build directory:

   set(NYUZI_COMPILER_ROOT <path to your nyuzi dir>/tools/NyuziToolchain/build CACHE PATH "Path to nyuzi compiler install directory.")

If not, you could try changing the install path in the LLVM cmakelists to install in your user directory (in tools/NyuziToolchain/CMakeLists.txt, change CMAKE_INSTALL_PREFIX), then pointing NYUZI_COMPILER_ROOT to that.

jbush001, 2020-01-05:
I'm closing this for now, as it appears to be a configuration problem on the host system and not specific to this project.

closed 2020-01-05 by jbush001